SolidSoils/Arduino #5: Add socket interface for Arduinos using ESP-8266 serial-WiFi boards?

ShadGates, 2015-05-01:
I just found your library, and haven't browsed your code yet to see how difficult it would be...

I would like ot propose adding a TCP/IP socket interface to your library, enabling it to talk to Arduinos using the serial-to-wifi modules, like the ESP-8266, since these are very inexpensive, and easy to configure (I'm using several in Arduino projects already), and they present like any other serial-connection, once configured (all of the WiFi work happens on the ESP module).

Thanks for the library!

SolidSoils, 2015-05-02:
Central piece of the library is the ArduinoSession class. It needs an object implementing the ISerialConnection interface to do its work. So, I guess we would only have to design a connection class that uses a TCP/IP socket instead of the serial port.

Do you know of a C# example connecting to the ESP-8266? You are welcome to use and/or contribute to the library. When you think it is useful, please consider to give it a star.

SolidSoils, 2015-05-07:
Status update: earlier this week I started development of an 'IPSocketConnection' class supporting network-connected devices like the ESP-8266. Things are looking promissing!

ShadGates, 2015-05-07:
Excellent!
I do c# TCP/IP dev for a living (working for a credit-card processing company, talking to point-of-sale terminals out in the field), and was planning on writing an interface for you, but haven't had time to start yet.
Looking forward to seeing what you come up with!
Thanks,Shad