SolidSoils/Arduino #6: How do I set pin 13 - onboard led to high?

richardjharding, 2015-06-13:
Hi,

I've compiled the library and run the basic example code so am happy that I can connect to the arduino
but when I try to set pin 13 high with the following:

```
firmata.SetDigitalPinMode(13, PinMode.DigitalOutput);
firmata.SetDigitalPin(13, 1);
```

Nothing happens - when I try the firmata test utility [here](http://firmata.org/wiki/Main_Page) I can set the pin ok and it lights the onboard led - what am I missing?

Any ideas on how to debug?

SolidSoils, 2015-06-15:
Hi,

`SetDigitalPin()` sets an analog value on a PWM or Servo enabled pin. When you need to set a digital value on a digital port, you need method SetDigitalPort().

`SetDigitalPort(0, 32)` sets pin 6 of port 0 on an Arduino Uno. `SetDigitalPort(0, 0)` clears the pin (or actually all pins on the port).

The IFirmata protocol is designed to map closely to the Firmata protocol standard. Setting each pin individually would also be a bit inefficient, because it would require multiple commands to set all pins on a port where only one would do the job as well.

richardjharding, 2015-06-16:
Thanks for the explanation - so in practice when setting a single pin I need to read all the pins and merge my desired pin state with the current state of the other pins so as not to accidentally overwrite the current state of other pins on the same port?

SolidSoils, 2015-06-16:
Yes, this is also how Firmata on Arduino works. You cannot set a single bit/pin on a hardware register ('port') without affecting the other bits.

Googling such a feature brought me to this page on Github: [https://github.com/firmata/arduino/issues/102](https://github.com/firmata/arduino/issues/102 "add ability to set an individual digital pin value #102"). It's a discussion I initiated when I was designing this library. (Totally forgot about it, LOL.) At the end of the discussion Soundanalogous commented (a few weeks ago) he added the ability to set individual pin values recently.

My library is currently on version 2.3 of the Firmata protocol, but I hope to upgrade it to version 2.5 later this year.

SolidSoils, 2015-06-21:
Added an overload for `SetDigitalPin`:

    SetDigitalPin(int pin, bool value)

sets a pin that has been set in digital output mode.

closed 2015-06-21 by SolidSoils

kuklovod13, 2015-07-16:
SetDigitalPin(int pin, bool value)
is not working