SolidSoils/Arduino #11: Help documentation

Fin3, 2015-10-07:
Hi,

in windows 10 i'm not able to view the chm file, i can see the tree of the documentation but when i click a section nothing happen.

I was trying to find out a solution to get working the I2C communication, when i try to send data nothing happens! I dunno why but without a documentation i can't find out a solution.

this is my code:

            for (int i = 0; i < 255; i++ )
                session.WriteI2C(0x20, new byte[] { Byte.Parse(i.ToString()) });

i've tryed to put up ALL possibile combinations, but nothing happens (u've wired a PCF8574 with a common double relay board, so in any case i must heard the coil switching)...

Can i have a documentation working or i am doing something wrong opening it?

Thank you in advice for your reply, and compliments for the great work!

Fin3

SolidSoils, 2015-10-07:
Looks like you need a CHM reader. Could this help you out? http://chmreader.windows10appstore.net/win10apps.html

Can you connect to the board and send/receive data? Can you share your Arduino sketch?

Fin3, 2015-10-07:
Hello again,
ok i've finally opened the chm file, i was hoping to find more informations about the method "WriteI2C", unfortunately there isn't.
I will try again with other combinations.
If this can help you to investigate what's wrong, when i upload the sketch below, everything goes as expected, and the coil is blinking on and off every 1500 seconds.
#include<Wire.h>void setup() {  // put your setup code here, to run once:Wire.begin();}
void loop() {  // put your main code here, to run repeatedly:  Wire.beginTransmission(32);  Wire.write(0);  Wire.endTransmission();  delay(1500);  Wire.beginTransmission(32);  Wire.write(1);  Wire.endTransmission();  delay(1500);}
In any case, thanks for the time u dedicate to my issue.
Fin3

Date: Wed, 7 Oct 2015 03:57:57 -0700

SolidSoils, 2015-10-07:
On your Arduino board you need a sketch implementing the Firmata protocol, including I2C. StandardFirmata is a sketch fully supporting this protocol. You can find it here:
https://github.com/firmata/arduino/blob/master/examples/StandardFirmata/StandardFirmata.ino

Fin3, 2015-10-07:
Dear SolidSoils,

StandardFirmata is just the sketch i'm already using.

SolidSoils, 2015-10-07:
The sketch you shared only seems to include the Wire library, though. Am I missing something?

Fin3, 2015-10-07:
Maybe i wasn't so clear with my post.

When i used the sketch below:

				#include<Wire.h>
				void setup() {
				// put your setup code here, to run once:
				Wire.begin();
				}

				void loop() {
				// put your main code here, to run repeatedly:
				Wire.beginTransmission(32);
				Wire.write(0);
				Wire.endTransmission();
				delay(1500);
				Wire.beginTransmission(32);
				Wire.write(1);
				Wire.endTransmission();
				delay(1500);
				}

Things works well, and my relay blink. I used this sketch only to check that everything goes ok.

Usually I use the StandardFirmata sketch uploaded into arduino, in combination of your library with this code in the client:

        private void button1_Click(object sender, EventArgs e)
        {
            Console.WriteLine("Started.");
            var connection = new EnhancedSerialConnection("COM4", SerialBaudRate.Bps_57600);            
            var session = new ArduinoSession(connection, timeOut: 250);

            session.SetDigitalPinMode(18, PinMode.I2C);
            session.SetDigitalPinMode(19, PinMode.I2C);

            session.WriteI2C(0x20, 0);
            session.ReadI2COnce(0x20, 0);
            Thread.Sleep(500);

            connection.Close();

            Console.ReadLine();
            Console.WriteLine("Ready.");
        }

so, with the pressing of a button (button1) in my UI, i try to replicate what my example sketch does, but nothing happens. I've tried with session.WriteI2C(0x20, 0); with byte[] from 0 to 255.

Hope this time to had clarify my experiment.

SolidSoils, 2015-10-07:
Thank you for your explanation. I am not too familiar with the Wire library, hence the misunderstanding.

The SolidSoils4Arduino library implements the Firmata protocol as described here: http://firmata.org/wiki/Protocol

Method `WriteI2C` sends a packet of bytes to Arduino as described on this page with the title "I2C read/write request". I cannot tell if address 0x20 you are using is correct, but in your event handler only a 0 value is written to it. Assuming this switches a relay off, do you have code in place switching it on?

Fin3, 2015-10-07:
Hello again SolidSoils,

i can't understand if is your code not working or what is going on in my code... Anyway now the line of code session.WriteI2C(0x20, 0); work, but with a little trick before...

i try to explain what i've done:

Using your code "as is" i can't figure out on how to write i2c commands, so i search a little in the StandardFirmata sketch code, where i found a global boolean variable named "isI2CEnabled" that is for setup setted as false.
If this variable is setted as false, nothing happen when the "WriteI2C" method of your library is called because of a if(isI2CEnabled) at the top of all I2C functions in the StandardFirmata sketch.

The way to enable I2C in the sketch seems to call a funcion named "enableI2CPins()", that is it self called inside the SYSEX-BASED commands section (method sysexCallback), switching in the command "I2C_CONFIG".

Doing the trick with the code below the job work well, without putting hands on your code: 

            var connection = new EnhancedSerialConnection("COM4", SerialBaudRate.Bps_57600);

            var session = new ArduinoSession(connection, timeOut: 250);
            session.MessageReceived += session_OnMessageReceived;

            session.SetDigitalPinMode(18, PinMode.I2C);
            session.SetDigitalPinMode(19, PinMode.I2C);

            byte[] command = new byte[4];
            command[0] = 0xF0;
            command[1] = 0x78;
            command[2] = 0x00;
            command[3] = 0xF7;

            connection.Write(command, 0, command.Length);

            session.WriteI2C(0x20, Byte.Parse(txtbPort.Text));

            try
            {
                //session.GetI2CReply(0x20, 0);
            }
            catch (Exception exc)
            {
                Console.WriteLine(exc.Message);
            }
            //Thread.Sleep(1500);

            //session.ResetBoard();
            //session.Dispose();
            connection.Close();

The only thing that is making me crazy is that the function "GetI2CReply" (that you can see as commented in the code above) is calling the REQUEST and not the REPLY SYSEX-BASED command, and happens that the code crash when messages are appended returning "Wait condition for I2CReply message timed out." and from now on any of the WriteI2C methods again called do not work anymore, i must restart (by powering off) the arduino.

I'm trying now to figure out on what's wrong... I will let you know :)

Please let me know if i'm doing something wrong... If there is a better way i haven't seen.

Thank you!!

Fin3, 2015-10-07:
Sorry, 

				session.WriteI2C(0x20, Byte.Parse(txtbPort.Text));

i've not mentioned that for switching the relay on and off, i'm currently using a TextBox where i put "0" (switch the relay on) or "1" (switch the relay off) and pressing the button.

The i2c expander work with a reverse logic, 0 = on, 1 = off. dunno why :)

Fin3, 2015-10-07:
Ok....

If i use the Async process, it work well

session.GetI2CReplyAsync(0x20, 1);

Dunno what if is not async!

Fin3, 2015-10-07:
The session.GetI2CReplyAsync() doesn't work anyway

SolidSoils, 2015-10-07:
Method `GetI2cReply` performs a synchronous I2C read operation. It sends an I2C read request and expects to receive an I2C reply message from the Arduino board. I do not know why it does not work in your scenario. It looks as if it is not handled as expected by the Standard Firmata sketch.

I2C reply messages are only sent by the Arduino board. With method `ReadI2CContinuous` you can request the board to continuously send these type of messages.

The board is then expected to return a continuous stream of I2C_REPLY messages at an interval which can be set using the `SetI2CReadInterval(Int32)` method. Received I2C_REPLY messages trigger the `I2CReplyReceived` event. The data are served in the `I2CEventArgs`'s Value property as an `I2CReply` object.

The board can be stopped sending I2C_REPLY messages by calling the `StopI2CReading()` method.

Fin3, 2015-10-07:
Thanks for your help.

I don't know whats wrong, but the GetI2cReply still not work, when ReadI2CContinuous work as expected!

Thanks again.

SolidSoils, 2015-10-07:
You're welcom, glad I could help!

closed 2015-10-07 by SolidSoils