SolidSoils/Arduino #14: Unable to get 'GetPinState' from Analog input to return value

virtq, 2016-06-30:
I'm trying to get a value from an analog input. I have tested with a standard arduino sketch together with a potentiometer and I'm getting values between 0 - 1023 as expected. However when I try to do the same using your library I get the value 0 every time. It can recognize values from digital ports though. Maybe I don't understand how to use the library correctly. Do I need to set AnalogReportMode to true for the channel?

Code:
```
private void ReadButton_Click(object sender, EventArgs e)
{
            //initialize connection to Arduino
            var connection = new SerialConnection("COM5", SerialBaudRate.Bps_57600);
            var session = new ArduinoSession(connection, timeOut: 250);
            
            session.SetDigitalPinMode(16, PinMode.AnalogInput);
            PinState ps = session.GetPinState(16);
            MessageBox.Show(ps.Value.ToString() + " " + ps.Mode.ToString() + " " + ps.PinNumber.ToString());
            Thread.Sleep(500);
            connection.Close();
}
```

SolidSoils, 2017-12-31:
Method `SetAnalogReportMode` is used to command an Arduino board to send a continuous stream of sampled values. `GetPinState` retrieves a pin's single value.

I cannot reproduce the issue.

closed 2017-12-31 by SolidSoils

Oluntridao, 2018-02-28:
The same think happens to me. I always receve 0 value when i execute getPinState.

Angelusvetus, 2019-11-27:
I also have similar problems. I see a tremendous number of posts with the same questions. I feel the solidsoils library is well written. However, to date, I have not been able to find anyone providing information on using the analog side of solidsoils. All responses are vague and without example. There are a multitude of videos and text articles on the digital side. However, the digital side is rarley in question.

It would be very helpful if someone could provide a short example in C#.NET that reads an analog value from Analog input pin A0 on the Arduino UNO. This would be most helpful. The internet and myself would be most grateful.