SolidSoils/Arduino #19: Restart arduino

facetime88, 2017-12-28:
When app start running, the app cannot communicate with arduino, cannot send message neither receive any message. But when the app still running and arduino is reset manually, the app start to receiving message.

In arduino IDE, when we start serial monitor, this will restart arduino. But this not happen in this case. How do I restart arduino everytime the app is running?

facetime88, 2017-12-28:
I am using arduino due anyway

facetime88, 2017-12-29:
It seems the default value when app running for _base.DtrEnable_ is set to false. I add _base.DtrEnable = true;_ before base.Open(), and everytime app is starting, the arduino will also restarted.

I propose you also add DtrEnable as property.
Thank you

SolidSoils, 2017-12-31:
`EnhancedSerialConnection` is a subclass of `SerialPort`. Therefore there is no need to add property `DtrEnable`. You can use it like this:

```csharp
var connection = new EnhancedSerialConnection("COM3", SerialBaudRate.Bps_57600);
connection.DtrEnable = true;
var session = new ArduinoSession(connection);
```

facetime88, 2018-01-03:
It said 'DtrEnable' is not a member of ISerialConnection.

closed 2018-04-30 by SolidSoils