SolidSoils/Arduino #22: GetPinState Just Return 0

Oluntridao, 2018-02-27:
Hello, i'm trying to use this library in my project, but i cannot do it.

My code:
arduinoSession.SetDigitalPinMode(port.Numeric, Solid.Arduino.Firmata.PinMode.DigitalInput);
arduinoSession.SetDigitalReportMode(port.Numeric, true);

Thread thread = new Thread(() =>
{
   bool teste = false;
   while(!teste)
   {
       Console.WriteLine(arduinoSession.GetPinState(port.Numeric).Value.ToString());
   }
                
});

thread.Start();

Always console shows me 0 value.

skpotocny, 2018-09-07:
I am also having this same problem.  If I bind to DigitalStateReceived, I will get the valid PinState in the callback, but just calling GetPinState is not working for me.  I am assuming that GetPinState will block until the current pin state is accessed and returned.  Is this not correct?

            ard.SetDigitalPinMode(9, PinMode.DigitalInput);

            while (!Console.KeyAvailable)
            {
                ps = ard.GetPinState(9);
                Console.Write("{0}", ps.Value);
            }

skpotocny, 2018-09-12:
I found the answer - The Firmata Pin State Query message is not what I expected:

"The pin state is any data written to the pin (it is important to note that pin state != pin value). For output modes (digital output, PWM, and Servo), the state is any value that has been previously written to the pin. For input modes, typically the state is zero. However, for digital inputs, the state is the status of the pull-up resistor which is 1 if enabled, 0 if disabled."

closed 2018-10-11 by SolidSoils