SolidSoils/Arduino #23: Documentation unsupported error

evilC, 2018-03-01:
Upon forking and cloning this repo, when I open in visual studio 2017, I get the following message:  

[image]

Attached is the upgrade log
[UpgradeLog.zip](https://github.com/SolidSoils/Arduino/files/1772378/UpgradeLog.zip)

SolidSoils, 2018-03-04:
The documentation project can only be loaded when the SandCastle Help File Builder is installed on your computer. See https://github.com/EWSoftware/SHFB/.

evilC, 2018-03-04:
Do I need this?  
[image]

SolidSoils, 2018-03-04:
Yes, you do.

evilC, 2018-03-04:
OK, I got it to build.  
I take it that this just creates the class reference [found here](https://solidsoils.github.io/Arduino/html/3a2ac349-bf62-313a-9f20-4c7c3932b60a.htm)?

SolidSoils, 2018-03-04:
Exactly, it generates the reference documentation from the source code. With SandCastle though it is perfectly possible to create full fledged help files.

evilC, 2018-03-04:
Right, I suppose it is just taking the decorations and packaging it all up.  
So do you envisage putting all of the documentation in here?  
For example, do you think in the docs for `SetDigitalReportMode` should give some hint as to what a port is?  
That totally stumped me for a while, especially as when you search for `Arduino Firmata Port`, you don't get much that is helpful...  
Not looking to push all the work on you, I am happy to help out, I think the `Solid.Arduino.Run` project could do with some cleaning up, and would be happy to contribute new stuff.  
FYI, I can be found in the gitter [Firmata Arduino](https://gitter.im/firmata/arduino) channel or [on Discord](https://discord.gg/MmnhQYQ) if you wanna take this to chat.

SolidSoils, 2018-03-05:
I think it's a convenient way of documenting the software. For `SetDigitalReportMode` some extra guidance can indeed be added. Projects Solid.Arduino.Monitor and Solid.Arduino.Run were intended as trials, but I left them in an unfinished state. Maybe they should be removed. And, of course, your contributions are welcome here!

closed 2018-04-30 by SolidSoils