SolidSoils/Arduino #25: IServiceLocator not referenced in Solid.Arduino.Monitor

evilC, 2018-03-04:
ViewModelLocator.cs, Line 32  
`ServiceLocator.SetLocatorProvider(() => (IServiceLocator)SimpleIoc.Default);`

```
Error	CS0012	The type 'IServiceLocator' is defined in an assembly that is not referenced. You must add a reference to assembly 'Microsoft.Practices.ServiceLocation, Version=1.3.0.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35'.	Solid.Arduino.Monitor	Arduino\Solid.Arduino.Monitor\ViewModel\ViewModelLocator.cs	31	Active
```
So I add the [NuGet package](https://www.nuget.org/packages/CommonServiceLocation)  
Then I get another error  
```
Error	CS0104	'ServiceLocator' is an ambiguous reference between 'CommonServiceLocator.ServiceLocator' and 'Microsoft.Practices.ServiceLocation.ServiceLocator'	Solid.Arduino.Monitor	Arduino\Solid.Arduino.Monitor\ViewModel\ViewModelLocator.cs	32	Active
```  

So then I try a different approach.  
In NuGet Packages, I see that there is an update available for `CommonServiceLocator`.  I install the update, but I get the following dialog:   
[image]

evilC, 2018-03-04:
Pushing on, I click yes, then try to build - 2 errors, 1 solved by adding `using CommonServiceLocator;`, the other is in App.xaml, there appears to be an unmatched XML tag `<ResourceDictionary />`, so I delete it.  

The app finally builds and loads, but clicking Connect does not seem to do anything - trying to work out why now...

floppydisk525, 2018-03-10:
I preface my comment that I'm not a good programmer so this might not be helpful.  But, it seems like it might.  

The MVVM Light service locator has changed slightly per Laurent B at the bottom of the page here:  
http://www.mvvmlight.net/std10

I copied the pertinent text:

Known issues
Unfortunately, it was not possible to port MVVM Light as is. One particular interface, IServiceProvider, which is used by the SimpleIoc component, is not available in .NET Standard 1.0. After some thinking and consulting with the .NET team, I decided to remove this interface from MVVM Light .NET edition. This requires some changes in your application code, as described below.

Note: If you continue to use the portable class library version of MVVM Light, nothing changes. This only concerns the .NET Standard 1.0 version of MVVM Light, contained in the Nuget package called MvvmLightLibsStd10.

The IServiceProvider interface was used by the ServiceLocator class (from the CommonServiceLocator Nuget package). This class was used historically to allow developers to easily switch from one IOC container to another without having to modify multiple locations. This sounded like a great idea but in recent years, this strategy raised a few concerns. It added a dependency on an external library (CommonServiceLocator) which was not actively maintained. When new platforms (such as Windows 8.1 or Windows 10) were released, porting the CommonServiceLocator took some time, which in turn delayed the release of the dependent libraries.

In MVVM Light for .NET Standard 1.0, I decided to remove this dependency completely. As a result, it means that the ServiceLocator class is not available anymore.

For an existing application, this means the following changes (old code commented out).

VIEWMODELLOCATOR.CS

using section:

`using GalaSoft.MvvmLight;
using GalaSoft.MvvmLight.Ioc;
using GalaSoft.MvvmLight.Views;
// OLD using Microsoft.Practices.ServiceLocation;
using MvvmLightWithPcl.Model;`

constructor:
`static ViewModelLocator()
{
    // OLD ServiceLocator.SetLocatorProvider(() => SimpleIoc.Default);
 
    // Rest of the constructor unchanged
}`

MAINVIEWMODEL.CS AND OTHER LOCATIONS:

In the rest of the code, you need to remove every mention to ServiceLocator.Current and replace with SimpleIoc.Default. For example:
// OLD var nav = ServiceLocator.Current.GetInstance<INavigationService>();
// NEW
var nav = SimpleIoc.Default.GetInstance<INavigationService>();

jeroenwalter, 2019-04-02:
I think you can ignore this entire Monitor project, as it has no code for actually connecting to an Arduino. Look in the MainViewModel class, it is empty.
Also the Connect button has no binding to a command in MainViewModel.
Also, MVVM light and its SimpleIoc is not necessary for such a small project, just instantiate the MainViewModel as the datacontext in the XAML file itself.

To the devs: Are there plans to develop this Monitor project any further?

SolidSoils, 2019-04-23:
That's correct. The Monitor project was just a quick setup for a WPF app, but due to lack of time I did not complete it.

closed 2019-04-23 by SolidSoils

jeroenwalter, 2019-04-23:
I did create a very simple monitor program for my own purpose. I think I can refit it easily to replace the Monitor project.