SolidSoils/Arduino #26: io exception in site test example

jose2k72, 2018-04-26:
Hello, the same thing is happening to me as reference the problem # 15. This problem was closed without a concrete answer.
I am using an Arduino Leonardo
In my case my computer connects to the arduino, when I put a comment on the line that does the automatic search of the arduino and use, for example, the following change, it works for me.

// ISerialConnection connection = EnhancedSerialConnection.Find ();
ISerialConnection connection = new EnhancedSerialConnection ("COM3", SerialBaudRate.Bps_115200);

I write on pin 13 to make tests and everything goes as it should be.

I have also detected that something similar happens to me in the following parts of the code

var firmware = session.GetFirmware ();

var protocolVersion = session.GetProtocolVersion ();

BoardCapability cap = session.GetBoardCapability ();

I have done a bit of debugging in the code and I have seen that it gives me an exception of type System.IO.IOException

I put the detail of the exception here in case it can be helpful

=====================================================

System.IO.IOException was unhandled
  HResult=-2147024894
  Message=The system cannot find the file specified.

  Source=Solid.Arduino
  StackTrace:
       at Solid.Arduino.SerialPortFixer.ThrowIoException() in D:\PYTHON.SRC\IMRE\vs_src\solidsoils\Solid.Arduino\SerialPortFixer.cs:line 137
       at Solid.Arduino.SerialPortFixer.Initialize(String portName) in D:\PYTHON.SRC\IMRE\vs_src\solidsoils\Solid.Arduino\SerialPortFixer.cs:line 90
       at Solid.Arduino.EnhancedSerialPort.GetInitializedPortName(String portName) in D:\PYTHON.SRC\IMRE\vs_src\solidsoils\Solid.Arduino\EnhancedSerialPort.cs:line 104
       at Solid.Arduino.EnhancedSerialPort..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits) in D:\PYTHON.SRC\IMRE\vs_src\solidsoils\Solid.Arduino\EnhancedSerialPort.cs:line 97
       at Solid.Arduino.EnhancedSerialPort..ctor(String portName, Int32 baudRate) in D:\PYTHON.SRC\IMRE\vs_src\solidsoils\Solid.Arduino\EnhancedSerialPort.cs:line 79
       at Solid.Arduino.EnhancedSerialConnection..ctor(String portName, SerialBaudRate baudRate) in D:\PYTHON.SRC\IMRE\vs_src\solidsoils\Solid.Arduino\EnhancedSerialConnection.cs:line 32
       at Solid.Arduino.Run.Program.DisplayPortCapabilities() in D:\PYTHON.SRC\IMRE\vs_src\solidsoils\Solid.Arduino.Run\Program.cs:line 66
       at Solid.Arduino.Run.Program.Main(String[] args) in D:\PYTHON.SRC\IMRE\vs_src\solidsoils\Solid.Arduino.Run\Program.cs:line 24
       at System.AppDomain._nExecuteAssembly(RuntimeAssembly assembly, String[] args)
       at System.AppDomain.ExecuteAssembly(String assemblyFile, Evidence assemblySecurity, String[] args)
       at Microsoft.VisualStudio.HostingProcess.HostProc.RunUsersAssembly()
       at System.Threading.ThreadHelper.ThreadStart_Context(Object state)
       at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
       at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
       at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
       at System.Threading.ThreadHelper.ThreadStart()
  InnerException:

SolidSoils, 2018-04-30:
Class SerialPortFixer uses a few Windows libraries which may be not available on your operating system. What OS are you using?

jose2k72, 2018-04-30:
I am currently using windows 8 and windows 10 with visual studio 2015 enterprise edition

jose2k72, 2018-05-04:
So? What I do? Please give me a hint to follow. I really need to use a decent library and it seems to me that yours is. I really do not have much desire or time to study the protocol and implement my own.