SolidSoils/Arduino #27: EnhancedSerialConnection.Find(); Results In Timeout When Arduino Is Connected

Brittank88, 2018-09-12:
As the title suggests, `EnhancedSerialConnection.Find();` results in a timeout error when an Arduino device is connected to one of my ports. If there is no device connected, the connection is null meaning it is indeed picking up the device, but timing out before whatever data exchange occurs can fully take place.

Is there any way for me to disable the timeout of `EnhancedSerialConnection.Find();`?

skpotocny, 2018-09-13:
I had a similar problem working with the embedded Leonardo on a LattePanda SBC.  I found that setting DtrEnable after creating the EnhancedSerialConnection cleared this up.  I found this suggestion on the issue 19 message trail.

Brittank88, 2018-09-16:
Thanks for your advice!

Unfortunately I'm unable to test this as the project involving it concluded (and there were other larger issues present anyway).

Hopefully others come across this if they're having the same issue.

closed 2018-10-11 by SolidSoils