SolidSoils/Arduino #29: Severe performance impact of Debug.Write in SerialDataReceived

jeroenwalter, 2019-04-05:
I have an Arduino sketch that produces about 20 sysex commands per second, each about 30 bytes long.
Writing every single byte to the Debug output has a severe impact on the performance of ArduinoSession. This causes serial data to be queued and processed seconds after they have been received from the Arduino.
I can understand the need to view the raw serial data, but it would be nice if this can be conditionally be turned off.

SolidSoils, 2019-04-23:
I will replace the DEBUG verb with another word.

jeroenwalter, 2019-04-26:
I have addressed this in my branch by creating a simple logger interface and using that instead of Debug.WriteLine(). Then in the program you can implement your own logger class, for instance one the outputs to console or debug window. You then have more control over performance.

SolidSoils, 2021-10-15:
Debug.WriteLine statements wrapped in TRACE constants.

closed 2021-10-15 by SolidSoils