SolidSoils/Arduino #30: .net core 3.0 support for Raspberry Pi

jeroenwalter, 2019-04-21:
Just to let you know, I'm working on .net core 3.0 support for this library, so I can get firmata to work on a Raspberry Pi.
I've also already added support for custom sysex messages. Still need to make some unit tests for that though.
My fork: [https://github.com/jeroenwalter/Arduino/tree/firmata_msgsize](url)

SolidSoils, 2019-04-23:
That's nice!
I've been thinking of porting this library to .net core myself, but had no time to pick this up yet. I have been working on support for TCP/IP and UDP connections, but have no working version yet. It would be great to be able to connect to ESP-8266 boards though.
It's all about time...

jeroenwalter, 2019-04-26:
I'm currently separating the Firmata related classes from the Serial connection classes in my fork.
This would enable other connection types, such as sockets, to be used.
I've done a lot of work rearranging files into separate projects, but I'm not there yet. 
I still need to fix the test projects (they don't compile now) and I want to make some projects multi-targeted, so I have 1 project for both netcore and net47/mono. 
You can check my progress on my fork, I've just pushed my changes.
The command line Run projects work.

I have a question about the EnhancedSerialConnection, EnhancedSerialPort and SerialPortFixer classes. Do you think they are still necessary for mono? I would rather remove them if they have no use anymore.

SolidSoils, 2021-10-15:
Library has been upgraded to .NET Standard 2.0

closed 2021-10-15 by SolidSoils