SolidSoils/Arduino #32: System.NotImplementedException: 'Commandbyte 8D is not implemented.'

jgmdavies, 2019-07-09:
Anyone else get this unhandled exception when you restart after your program crashes or you terminate it?

It's easy enough to reset the Arduino and restart, but it would be better to handle it in the first place.

jeroenwalter, 2019-07-09:
I had that exception as well.

It seems that in C# upon restart or crash, that the usb/comm connection to the Arduino isn't closed, as the Arduino should normally reboot when a new connection is made. In C++ I've seen the correct behaviour, i.e. the Arduino reboots when a new connection is made. In C# in Windows this isn't the case, though on a Raspberry Pi the Arduino reboots as well. Maybe it's got something to do with running the C# program in the debugger.

As the Arduino isn't rebooted, it'll continue to send its firmata messages, even if noone is listening. So once your program connects again, the Arduino may be in the middle of a firmata command, while the ArduinoSession expects the start of a firmata command. Hence the firmata byte stream from the Arduino is out of sync with the firmata parser in ArduinoSession. This can also be caused by an incorrect baudrate, but that's likely not the case here when you restart your program.

So, how to handle this?

Unfortunately throwing an exception here is the worst one can do here (in ArduinoSession.cs).
As this exception is thrown in an event handler, in a random thread, this exception can only be caught in the global unhandled exception handler, which is not the place you want to be for this problem.
The best thing to do here is just not to throw the exception, but instead skip the received byte and wait for the ArduinoSession to sync with the firmata stream from the Arduino. 
That's what I did in my fork (which isn't ready for production yet...)
See https://github.com/jeroenwalter/Arduino/blob/netcore30/Solid.Arduino.Firmata/ArduinoSession.cs lines 1112 and 1124.
In the original source (https://github.com/SolidSoils/Arduino/blob/master/Solid.Arduino/ArduinoSession.cs) these are the lines 1052 and 1058.

But you'll have to compile the source yourself instead of using a nuget package.

jgmdavies, 2019-07-09:
Many thanks Jeroen!
If there's enough interest, maybe someone could release a fix (also on NuGet)   ;---)
Jim

zifeiniu, 2021-10-10:
session.SetAnalogReportMode(1, true);
    session.AnalogStateReceived += Session_AnalogStateReceived;

SolidSoils, 2021-10-15:
Fixed. NotImplementExceptions no longer throwed in event handling code.

closed 2021-10-15 by SolidSoils

zoopyserg, 2021-11-04:
I had this problem (I have an Analog Input connected to a temperature sensor and have a TemperatureChangeEventHandler).
My solution was just to call ````firmata.ResetBoard();```` right before I close the connection.
Which seems to have solved most of my problems.