SolidSoils/Arduino #33: SolidSoils Arduino Firmata Analog Methods

Angelusvetus, 2019-11-27:
I am having trouble reading a value on pin A0(14) of an Arduino UNO. 
 I see a tremendous number of posts with people having the same issues. I feel the SolidSoils library is well written. However, to date, I have not been able to find anyone providing information on using the analog side of this library. All responses are vague and without example. There are a multitude of videos and text articles on the digital side. However, the digital side is not my issue, it is intuitive and I have no problem getting it to work.

It would be very helpful if someone could provide a short example in C#.NET that reads an analog value from Analog input pin A0 on the Arduino UNO using Firmata. This would be most helpful. The internet and myself would be most grateful.

jacobussystems, 2019-12-09:
@Angelusvetus 

Using Standard Firmata with an Arduino Due, I had no luck at all using 'GetPinState' to work on either analog or digital pins - does anyone have more luck or more info on this?

But I've been successfully using the ReportMode mechanism, wherein you subscribe to changes in analog state, e.g.

	m_ArduinoSession.SetAnalogReportMode(chan, state);
	m_AnalogStateMonitor = m_ArduinoSession.CreateAnalogStateMonitor();
	var unsubscriber = m_AnalogStateMonitor.Subscribe(this));

and an event handler like this:

		// Implementation of: IObserver<AnalogState>.OnNext.
		public void OnNext(AnalogState state)
		{
			// The new analog value is in 'state'Level'.
		}

Sorry I haven't got time to do a complete example, but maybe you can piece it together from the above?

HTH,
Jim

Angelusvetus, 2019-12-13:
It is late here, but I will take a closer look at this in the morning. It looks promising, and I do appreciate you taking the time to share this information. 

Again, Thank You.

Best Regards,

Rhett

Angelusvetus, 2020-01-04:
@jacobussystems 

Hi Jim,

I looked at the snippet you posted. I was not able to get it to work. I am sure it is good code, but I should have mentioned in my original post that I am entry level in programming. My original goal was just to have a ProgressBar control display the analog input (A0) value (I.e. myProgressBar.Value = myArduinoAnalogInputPin's value). I think this should be easy to do, but after many hours of trying to get this to happen, I have been unsuccessful at getting my head wrapped around what is and what should be happening here. Everything in this solidsoils library has a tutorial for using with C# out there with one exception... anything analog.

I was having trouble with subscribing and the event handler. I could not get the .OnNext to show up in the intellisense and the "this" keyword threw errors on converting between form and AnalogState.
I wish I knew more about this to give better information.

If you could give a brief description of the sequence of events that should take place for this, that would be greatly appreciated. If you do not have time to mess with it, I understand also. 

Thank You Sir.

Rhett

jacobussystems, 2020-01-04:
@ Angelusvetus

Hi Rhett,
Just a quickie for now re. the 'OnNext' and 'this' problems.  You need to also add the relevant interfaces to the class definition (the class where you have the 'OnNext' method), e,.g.

```
	public class YourFirmataDevice : YourDevice, IObserver<DigitalPortState>, IObserver<AnalogState>
	{

```
I'm assuming that your class derives from another class of yours called 'YourDevice',, otherwise it's just:

```
	public class YourFirmataDevice : IObserver<DigitalPortState>, IObserver<AnalogState>
	{

```
Hope this helps a bit!
Jim

Angelusvetus, 2020-01-04:
@jacobussystems 

Hi Jim,

That did help tremendously! 
I was able to get the state of one analog input to be displayed in a ProgressBar on my form. It is stable and responsive. Now, I will try to add the other five analog inputs and five more ProgressBars to reflect the states of all the analog inputs on my Arduino board onto my form displayed on my PC. I am not sure how to differentiate the inputs/channels with only one .OnNext(AnalogState state). That is my next mountain to climb.

I very much appreciate the help you have provided Sir.

Best Regards,

Rhett

jacobussystems, 2020-01-04:
That's great.

Check out: state.Channel !

And data in state.Level.

Angelusvetus, 2020-01-04:
@jacobussystems 

Hi Jim,

You are a Genius!
I am almost ashamed to admit how many hours/days I spent with no real success.
The pieces of information you provided paved the way to success!
More importantly, I now have a much better understanding of how the software works with the hardware.

I do appreciate and thank you for the information and help you provide. 

Again Thank You Sir!

Best Regards,

Rhett

jacobussystems, 2020-01-05:
My pleasure!

I wish I was a genius...

If you're in that 'OnNext' method in Visual Studio, you can just type '**state.**' and a list of state's members should popup (Intellisense).

Best,
Jim

closed 2020-01-06 by Angelusvetus

zoopyserg, 2021-10-13:
Everyone here is way too polite to people who don't finish the job they've started. 

If you post an answer, just post full files, and not "add this line" and a couple of pagedowns later "add this line too".

zoopyserg, 2021-10-13:
So many "geniuses" and two years later after the initial response, there is no code example. I hate open-source projects.

zoopyserg, 2021-10-14:
This file from the SolidSolis repo contains a full working example of both Analog and Digital signal processing:

````URL
https://github.com/SolidSoils/Arduino/blob/master/Solid.Arduino.Run/Program.cs
````

Everything that starts with this:
````C#
static void SimpelTest
````
And until the end of the file.

But out of the box, it is also incomplete (sets up the callbacks, but starts listening for only digital pins).
To get it to work for analog pins, people need to toss in this line somewhere inside ````static void SimpelTest````:

````C#
firmata.SetAnalogReportMode(5, true);
````

Which then makes that code magically trigger the callback ````private void Session_OnAnalogStateReceived```` which contains the correct recording of an analog pin.

After that I need to solve the problems with parallel threads and busy open Arduino connections which is still work in progress. 

This is the answer I was looking for.
I'm a hero here.