SolidSoils/Arduino #35: Migrate to netstandard2.0

AngelMunoz, 2020-04-18:
Hey I'm getting an Arduino soon and I'm pretty much new to the Arduino space, but I was looking for options on .net it seems so far that you have the best-supported library so far even Microsoft doesn't seem to be doing much about Arduino, (Raspberry PI is another story though)

I wanted to see if you are interested in migrating from full .net framework to netstandard2.0
I gave myself a try at doing it on my fork primarily the library and the unit tests project (I did try to update version on the other projects, but no luck so far building the Monitor) and the projects build fine I ran the tests and it seems there are only 7 tests failing from the testing project, I can't get the Integrations test to run

but my guess is that I don't have (yet) an Arduino connected to my PC and that's making the tests fail
[image]

I know there's some talk about something similar in #30 perhaps we could coordinate
if you are interested let me know 😁 you can give my fork a test run and see if it's worth the effort

SolidSoils, 2020-04-20:
It has been a long time ago I committed my last update to this project and I feel it is time to add some updates. .Net Standard would be a nice target. Also want to implement some other features, e.g. TC/IP support is high on my list.

Don't bother much about the Monitor project. Years ago I intended to create a demo WPF program, but did not took it further.

AngelMunoz, 2020-04-21:
>Don't bother much about the Monitor project. Years ago I intended to create a demo WPF program, but did not took it further.

Sounds good! I'll take it away then.

 I managed to build the docs as well

[image]

as soon as I get my Arduino I'll try to run the tests again, so far everything builds and tests run fine (the ones that don't seem to require an Arduino to be able to be tested)

jeroenwalter, 2020-06-05:
You may try my .net core 3.0 fork.
https://github.com/jeroenwalter/Arduino/tree/netcore30
I haven't touched it in a while, but I'm now revisiting it, as I now need to use it on a Raspberry Pi.