SolidSoils/Arduino #38: Sysex Command

Psydis, 2020-08-05:
Is there any methods as "sendSysex" to customize protocol? SendStringData doesn't work fine in my project and I'd like to manage commands from SolidSoils or I fault to use that method.

Thanks

jeroenwalter, 2020-08-05:
If you want to send your own custom sysex messages, then you'll have to write your own method to fill in your sysex message and send it to the Arduino.

Take a look at the methods SendSysExCommand, SetSamplingInterval or SendStringData https://github.com/SolidSoils/Arduino/blob/master/Solid.Arduino/ArduinoSession.cs for good examples.

Basically, fill an array with your sysex content and write it to the connection. It's not that hard.

jeroenwalter, 2020-08-05:
In my fork for .netcore I have made method SendSysEx and SendSysExWithReply that you maybe can use as inspiration.
https://github.com/jeroenwalter/Arduino/blob/netcore30/Solid.Arduino.Firmata/FirmataSession.cs

Psydis, 2020-08-05:
> In my fork for .netcore I have made method SendSysEx and SendSysExWithReply that you maybe can use as inspiration.
> https://github.com/jeroenwalter/Arduino/blob/netcore30/Solid.Arduino.Firmata/FirmataSession.cs

That's incredible ... I just try to compile your fork some minutes ago on Core 3.1!!! I'll let you know if works!!!
Thanks!

jeroenwalter, 2020-08-05:
It's already configured for netcore 3.1 :)

Psydis, 2020-08-05:
> It's already configured for netcore 3.1 :)

I know but i'm a maniac ... i love to recompile code, so i can understand it better!!!